FaderOrg/Fader25_board #4: Assembly sequence

mkosunen, 2020-02-26:
@FaderOrg/a-team 
Slide 14
I know Kosta used Sierra for the board house and I just remembered who he used for SMT assembly (Jeditron ... I've never used them) and for die-attach, he used Quik-Pak.  But there was some order ... I don't remember.  I think it was Jeditorn (front or back SMT) then Quik-Pak then Jeditron (back or front).

mkosunen, 2020-02-26:
@FaderOrg/a-team 

This is what I found from mo e-mail archives:
On 27/09/2018 22:42, Kosta Trotskovsky wrote:
> Julian Flores
> Jeditron Technologies Corporation
> 44137 Fremont Blvd.
> Fremont, CA 94538
> jeditronjf@hotmail.com <mailto:jeditronjf@hotmail.com>
> 408-802-6795 Mobile
> 510-226-1383 Office
> 
> I only contacted him over email, his response time is very quick
> (usually less than an hour).
> 
> To get a quote I've sent the gerber files, the BoM and a picture of the
> flip-chip location that another vendor (Quik-Pak) assembles. The process is:
> 1. Assemble the top components with Jeditron
> 2. Assemble the flip-chip with Quik-Pak
> 3. Assemble the thru-hole (headers and SMAs) and bottom components with
> Jeditron

gwright83, 2020-03-02:
Commit c0ba88f closed this.  I copied the information from Kosta to slide 15.

closed 2020-03-02 by gwright83