FaderOrg/Fader25_board #5: Assembly test plan

mkosunen, 2020-02-26:
@FaderOrg/a-team 
Slide 14: However, what is critically important on Slide 14 is where it says "Write up a test plan on the following slides".  The reason you need to think through that is that it makes a difference in the files I need to create for each vendors Request for Quote.  That is, you need to think about how/when during this assembly process you are going to test various things (especially the die attach).

gwright83, 2020-03-06:
Kosta, was there any testing of the power supply (top of board) components on your FADER board before die attach?  Was there any testing after die attach but before bottom side components and SMA connectors went on?

kostatro, 2020-03-12:
Here are the tests that I have done during the assembly process:

- For an empty board I tested that the power and ground planes are not shorted, and that the chip pads are shorted to the power and ground planes. 

- The board was sent for top components assembly.

- After the assembly of the top components I tested the LDOs and level shifters. You can connect the FPGA to the board and see that at the die side you have the correct voltages and scan signals. On the bottom I only had caps and a few ferrites, so they were not crucial for this test.

- The board was sent to die attach and for bottom-side/through-hole assembly. There were no tests between the die attach and the bottom-side/through-hole assembly.

gwright83, 2020-03-12:
Thank you!  That is just what I needed.

Best Wishes,

Greg

kostatro, 2020-03-12:
Sure! BTW, did you actually write this comment 7 days ago like it says on Github? I received an email only 5 hours ago...

mkosunen, 2020-03-12:
I edited the comment and I pinged youo with @kostatro about five hours a go :).

kostatro, 2020-03-12:
I see. Please ping me right away if needed :)

gwright83, 2020-03-13:
Hey Kosta, were all the bottom side components hand soldered, or was there a reflow step?  (What I am really asking is whether you specifically asked not to have the board with the die attached send back through a reflow oven.)

gwright83, 2020-03-13:
ping @kostatro Hey Kosta, were all the bottom side components hand soldered, or was there a reflow step? (What I am really asking is whether you specifically asked not to have the board with the die attached send back through a reflow oven.)

gwright83, 2020-03-13:
ping @mkosunen Are there any external points we could probe to know if the die attach worked?  For example, is there more than one place on the board we could probe the PWR or GND that were internally connected on the chip?  This would mostly check that the balls were connected.  We could also check that PWR/GND were not shorted, but if this were done with an ohmmeter, we might need a current limiting resistor in series with the probe.

gwright83, 2020-03-13:
I have the test plan pretty much done, except for two questions: 1) can the board go back through the reflow oven after die attach, and 2) are there any points on the board which the die attach shop could easily probe to have an indication that the die attach succeeded.  I'll close this once I have some idea what to do about the first, which seems to be more critical.

kostatro, 2020-03-14:
As far as I remember the soldering of the bottom components was done manually, like the through-hole SMA connectors and headers. There were only passives at the bottom and not that many of them, so it was ok. If you have many components at the bottom, or active components with complicated footprints, it might not be possible to do manually.

kostatro, 2020-03-14:
Regarding the test if the die attach succeeded, I just did a visual inspection with a microscope. When you look from the side on the soldered chip you can see at least that the bumps on the sides are soldered. It's a good idea to do some electrical testing, but I didn't want to use LDOs without decap (the caps which are at the bottom).

gwright83, 2020-03-16:
I have updated the test plan based on Kosta's feedback.  The back side components should all be hand attached and the board should not go through a reflow oven after die attach.  After die attach, visual inspection should be sufficient to detect gross errors.  I also specified the use of Isola 370HR as the circuit board material, since we ought to be  able to use any old FR4 type material -- Megtron 6 would be overkill.

I've checked the updated test plan into the repo and updated the submodule in the Fader25_board repo.

closed 2020-03-16 by gwright83

mkosunen, 2020-03-16:
@gwright83 To your previous question about checking the die attach: not really. Also measuring somthing from chip that is not povered properly can return basically anything (floating gates cause more or less random resistance between supplies).