FaderOrg/Fader25_board #7: Slide 15 information

mkosunen, 2020-02-26:
@FaderOrg/a-team 
Slide 15 (everything in red is incorrect b/c it's left-over text from this template file as was filled in by Woorham/Zhaokai for their Intel22ADC board.  

Please update with info for your project.  You need to make decisions on how many boards you want ordered, for starters.  Write down exactly what you want to order on this slide.  Communicating with vendors takes a long time.  I am not good at ESP and we don't have a lot of time, so please tell me what you want by putting everything down in writing in this file.

gwright83, 2020-02-27:
I am working on this now.  Marko, do you remember if FADER2 was die attach first, then SMT, or the other way around?

aflynn, 2020-02-27:
I definitely remember it was in the middle of going to Jeditron twice ...

gwright83, 2020-02-27:
It seems as if the FADER2 assembly procedure was SMT components -> flip die attach -> SMA (maybe other) connectors.  I will put that in the test plan.

aflynn, 2020-02-27:
Are you saying it was frontside SMT first, or backside SMT first?

gwright83, 2020-02-27:
I do not know if it was frontside or backside SMT first.  Marko, do you remember?

mkosunen, 2020-02-27:
@FaderOrg/a-team I asked Kosta about frontside/backside order. Waiting for the reply. For the order of assembly, see #7

gwright83, 2020-03-05:
Updated bump pitch, diameter and material.  This closes the issue with slide 15.

closed 2020-03-05 by gwright83