FaderOrg/Fader25_board #13: I2C operation

mkosunen, 2020-03-12:
Hi @kostatro, (FYI @gwright83, @aflynn  )
I had to add a I2C controlled block to the desing (copied the control of VDD_LINK to supply the second SERDES and edited the Altium schematic accordingly). Could you elaborate how the supply conrol actually is implemented. I mean what is the principle, or is it documented somewhere. (Sure I can study the schematic, but this helps understanding it)

Thanks

kostatro, 2020-03-12:
I don't have access to Altium... Can you attach a pdf or a screenshot of the supply control schematic that you ask about?

mkosunen, 2020-03-21:
@kostatro
https://github.com/FaderOrg/Fader25_board/blob/master/Board_plan/Figures/Supplies_Isense.png
https://github.com/FaderOrg/Fader25_board/blob/master/Board_plan/Figures/Supplies_LDO.png
https://github.com/FaderOrg/Fader25_board/blob/master/Board_plan/Figures/Supplies_LDO_full.png

kostatro, 2020-03-21:
@mkosunen Now I've noticed that I saved the schematics into a pdf, so I have all of them.

Isense is a current sensing amplifier. It takes two voltage inputs VDDX_REG (the LDO output) and VDDX_RS (after a series resistor of 0.11ohms), amplifies it and provides a digital output. Practically I don't think that I used these outputs at all, just measured the total current externally.

I definitely used I2C to control the potentiometer to set the LDO voltage.

The I2C control comes from the FPGA. The only thing that I did with these control signals on the PCB was level shifting.

Let me know if it helps.

mkosunen, 2020-03-21:
@kostatro  https://github.com/FaderOrg/Fader25_board/blob/master/Board_plan/Figures/Supplies_Isense.png, I think is the only place I have seen those I2C_SDA_I and I2C SCL_I signals,  Is it so that I2C is somehow addressable so that I can drive all units controlled by it with the same signal. How the address is set?

Basically my concern is, that as I added U33 just bycopying U32 in Supplies_Isense and renaming outputs. Should that work?

kostatro, 2020-03-21:
@mkosunen Oh, I see you question. Now, that won't work. This sense amp has two address inputs A0 and A1. These analog inputs set the address according to the attached table (from the MAX9612 datasheet). I used the 4 first addresses. So you should connect U33 to a different combination of A0 and A1, not GND and GND.

[image]

mkosunen, 2020-03-21:
@aflynn This is a heads up:One more update coming in. Soon.

mkosunen, 2020-03-21:
Closed by commit  063fd9aab5aea1e31c7 of Fader25_board

@kostatro This may be unnecessy, but could you take a cross-check-double-check  glance at 
https://github.com/FaderOrg/Fader25_board/blob/master/Board_plan/Figures/Supplies_Isense.png

(Although the configuration should work with the same addrerss too, as they ar driving similar circuits)

Thanks.

@aflynn I'm done with this.

closed 2020-03-21 by mkosunen

kostatro, 2020-03-22:
@mkosunen Looks good to me. Why are you saying that it will work for the same address? This address is for reading the data from the block, so I don't think that we can read from two blocks with the same address.