FaderOrg/Fader25_board #14: Which .PcbDoc file is the right one to use?

gwright83, 2020-04-29:
@kostatro 
Which .PcbDoc file is the right one to use?  There are two: fader2_blind_vias_pcb.PcbDoc and fader2_pcb.PcbDoc. Which one corresponds to the board that was actually made?

kostatro, 2020-04-29:
Hi Greg, the blind vias is the final one. We first manufactured a board with through-vias, and saw that they create non-uniformity so chip attachment was risky. We did not assemble this board, and manufactured a new one with blind vias and uniform surface. The cost was obviously higher.

gwright83, 2020-04-29:
Thank you Kosta!

Back when I was a grad student, there was a Russian post-doc whose
office was done the hall.  He had quite a thick accent and the slightly
crazed look of a particle physicist.  He would have put it this way:
"Now weeee knnnnoooooooowwwwwww!

Greg

kostatro, 2020-04-29:
Haha I can actually hear the accent! :)

closed 2020-04-29 by gwright83