FaderOrg/Fader25_board #16: Tiny little pads on under the Fader chip

gwright83, 2020-05-15:
@kostatro @mkosunen 

I believe we understand the answer to this question, but want to be completely sure. The pads on the Fader pcb to which the signal bumps on the chip attach do not have any wider "pad" area at the attachment point, so it just looks like a narrow width board trace that ends at the edge of the chip. We presume that this is because the trace itself is wide enough to reliably connect to the 80 um diameter balls on the chip.

The power and ground connections seem to be aggregated onto larger connection areas on the board, with several bumps connected to a single power/ground pad that contains a blind via to the appropriate power/ground plane within the board.

True or false?

kostatro, 2020-05-17:
True. For the chip periphery there was no room for wider pads. They can also degrade the high RF frequency performance. The chip assembly vendor did not have any issues with assembling to these small pads.

gwright83, 2020-05-17:
Thank you, Kosta!

gwright83, 2020-05-18:
Closing this issue.

closed 2020-05-18 by gwright83