FaderOrg/Fader25_board #17: Bump discrepancy check

mkosunen, 2020-05-16:
@caolen , @gwright83 
This isssue is to solve the bump pcb-schematic diecrepancies reported by Carl. I have checked the connectivity from the chip netlist.
Related to issue #12 

TOP
- [x] IREF_ANALOG (Floating bump, no connection on chip)
- [x] SERDES_SUPPLY_PROBE_0 (Floating bump, no connection on chip)
- [x] LINK_BM_RX_0_N (Floating bump, no connection on chip)
- [x] LINK_BM_RX_0_P (Floating bump, no connection on chip)
- [x] LINK_BM_TX_0_N (Floating bump, no connection on chip)
- [x] LINK_BM_TX_0_P (Floating bump, no connection on chip)
- [x] VDD_BM_0 (Floating bump, no connection on chip)

RIGHT
- [x] LINK_BM_RX_3_N (Floating bump, no connection on chip)
- [x] LINK_BM_RX_3_P (Floating bump, no connection on chip)
- [x] LINK_BM_TX_3_N (Floating bump, no connection on chip)
- [x] LINK_BM_TX_3_P (Floating bump, no connection on chip)
- [x] VDD_BM_3 (Floating bump, no connection on chip)
- [x] REF_LO_N  (Floating bump, no connection on chip)
- [x] REF_LO_P (Floating bump, no connection on chip)

- [x] VDD_LINK_CM1 CONNECTIVITY MODIFIED/ADDED FROM PREVIOUS VERSION
- [x] REF_SERDES_CM1_P CONNECTIVITY MODIFIED/ADDED FROM PREVIOUS VERSION
- [x] REF_SERDES_CM1_N CONNECTIVITY MODIFIED/ADDED FROM PREVIOUS VERSION

BOTTOM
- [x] LINK_BM_RX_2_N (Floating bump, no connection on chip)
- [x] LINK_BM_RX_2_P (Floating bump, no connection on chip)
- [x] LINK_BM_TX_2_N (Floating bump, no connection on chip)
- [x] LINK_BM_TX_2_P (Floating bump, no connection on chip)
- [x] SERDES_SUPPLY_PROBE_2 (Floating bump, no connection on chip)
- [x] VDD_BM_2 (Floating bump, no connection on chip)

LEFT
- [x] REF_DIG_N (Floating bump, no connection on chip)
- [x] REF_DIG_P (Floating bump, no connection on chip)
- [x] LINK_BM_RX_1_N (Floating bump, no connection on chip)
- [x] LINK_BM_RX_1_P (Floating bump, no connection on chip)
- [x] LINK_BM_TX_1_N (Floating bump, no connection on chip)
- [x] LINK_BM_TX_1_P (Floating bump, no connection on chip)
- [x] VDD_BM_1 (Floating bump, no connection on chip)

- [x] VDD_LINK_CM0 CONNECTIVITY MODIFIED/ADDED FROM PREVIOUS VERSION
- [x] REF_SERDES_CM0_P CONNECTIVITY MODIFIED/ADDED FROM PREVIOUS VERSION
- [x] REF_SERDES_CM0_N CONNECTIVITY MODIFIED/ADDED FROM PREVIOUS VERSION
- [x] LINK_CM_RX_0_P CONNECTIVITY MODIFIED/ADDED FROM PREVIOUS VERSION
- [x] LINK_CM_RX_0_N CONNECTIVITY MODIFIED/ADDED FROM PREVIOUS VERSION
- [x] LINK_CM_TX_0_P CONNECTIVITY MODIFIED/ADDED FROM PREVIOUS VERSION
- [x] LINK_CM_TX_0_N CONNECTIVITY MODIFIED/ADDED FROM PREVIOUS VERSION

mkosunen, 2020-05-16:
Let me know if further details are needed, or close the issue.

caolen, 2020-05-18:
Marko, thank you.  This information helps.
Carl

caolen, 2020-05-18:
Marko, it is now clear what our next steps will be, based on the information you sent.  Please close the issue.

Best regards,
Carl
carl.olen@nokia-bell-labs.com<mailto:carl.olen@nokia-bell-labs.com>
+1 908 516-0582
+1 614 406-5443 (mobile)

closed 2020-05-18 by mkosunen