FaderOrg/Fader25_board #18: Physical dimensions of the Fader chip

gwright83, 2020-05-26:
@bwrcbcr 
Is there a .spec file or other file giving the physical dimensions of the FADER IC?  Perhaps there was one common to the CRAFT project chips?

bwrcbcr, 2020-05-26:
Hi Greg,

We do not have a SPEC file; we tend to rely on the GDS as the golden
information.

Thanks,

Brian

gwright83, 2020-06-05:
@caolen 
Do we still need a file giving the physical dimensions of the chip?  If not, I will close this issue.

caolen, 2020-06-08:
Greg, yes.  I still need to get the actual step file for the chip.  The extruded 3D body I placed as a model is just a placeholder.
Thanks,
Carl

gwright83, 2020-06-08:
@caolen 
Could we use a GDSII file of the top of the chip instead?  The BWRC flow never generates a .STEP file.

caolen, 2020-06-08:
Greg, I’m not familiar with the GDSII format.  If someone knows how to convert that it to a *.STEP file that would work.  Or, if someone has the shape, dimensions, and tolerances of the chip in a drawing, I can update the extruded 3D body for a more accurate model.
Thanks,
Carl

gwright83, 2020-06-08:
@caolen 
I will ask Brian Richards at the BRWC for the dimensions.  I know that they were on one of the emails I reviewed with him when I was in Berkeley at the beginning of March.

bwrcbcr, 2020-06-08:
All,

The only physical information that we have on the chip is GDS and submission spreadsheets; I am not familiar with other formats.

-Brian
(From mobile)

caolen, 2020-06-08:
Brian, so, if I can just get the numbers & units of width, length, height and dimensions of the chamfer at pin 1, I can work with these.
Thanks,
Carl

bwrcbcr, 2020-06-08:
Hi Carl,

It appears that there is a misunderstanding about the FADER21 bare die;
there is no package that would have a pin 1 marker or a chamfer; the
orientation is solely identified by the asymmetric pattern of bumps on the
die.  Specialized equipment may be required for the flip-chip die attach.

Thanks,

Brian

caolen, 2020-06-08:
Brian, my apologies for misunderstanding.  The footprint for the chip includes a silkscreen overlay that suggests a “chamfer” at pin location “A1.”

Since there is no package that would have a pin 1 marker or a chamfer, I simply need the length, width and height of the chip package.  Please provide these dimensions.
Thanks,
Carl

[A close up of a logo  Description automatically generated]

gwright83, 2020-06-08:
@bwrcbcr 
Hi Brian, I think Carl just needs the bare die size for the physical model of the board (no "package").

gwright83, 2020-06-08:
@bwrcbcr 
I think the thickness of the TSMC 16FFC die was 775 um.  Is that true?

gwright83, 2020-06-09:
@caolen 
I will measure the previous FADER IC die; the chip size is the same as the FADER 2.5 part.  The thickness is 0.775 mm.  I'll close this issue when I send you the dimensions.