FaderOrg/Fader25_board #19: Stacked blind vias?

gwright83, 2020-06-03:
@kostatro 
@mkosunen 
There seem to be stacked blind vias in the IC attachment area.  Is this necessary, or could the buried vias be offset in the middle layers?  This would help ensure the flatness of the IC mounting area, since the blind vias would all be from layers 1 to 2.

From Pekka Holm's email on the question:

> Carl mentioned via type L1-L2 used below chip, but when we start this project the vias below chip was thru vias, from Top to Bottom, L1-L6. Is this the via type that we want? (picture below)
> 
[image]
>  
> 
> Or do we need to change the via technology below chip to use L1-L2, L2-L5, L5-L6 structure instead of thru via L1-L6. (picture below)
>   
[image]
> 
> Just one comment to this last item, power planes have been in Layer3 and Layer4 and they will stay there. No change on that. Only the way how these planes are defined in Altium tool is changed to more visual way, and as Carl said, the Altium tool layer type is now “Signal” for these plane layers L3, L4. And yes, this gives possibility to put traces on layers L3 and L4 if needed. Earlier ground (VSS) planes on Layer2 and Layer5 have been defined with using this Signal layer type too. No change on that.
> 
>  
> 
> BR
> 
> -Pekka-
> 

Please comment, so we can resolve this on our next phone call.

aflynn, 2020-06-03:
I had something similar on EagleX and Sierra changed my Thru vias to
stacked since I already had each of the other blind/buried vias in other
places on my board anyway.  It saves them a process step (no need to drill
and plate Thru vias at the end, if during their process, they've made all
necessary blind/buried vias anyway.

Also, Sierra is fine with stacked vias ... no need to switch from stacked
vias to staggered/dog-bone vias for them ... their process has equivalent
risk/no-risk for stacked as for staggered, so might as well use stacked,
takes up less space and you don't need to add dogbone traces on your power
layers.

-Anita

gwright83, 2020-06-03:
Thank you Anita and Kosta.  I had not realized that the stacked vias were done at the suggestion of Sierra.

aflynn, 2020-06-03:
It wasn't so much that Sierra was suggesting a "stacked via" for the sake
of a stacked via alone ... it's that the first rev that Kosta did put thru
vias between 2x2 quartets of pads ... where those Thru vias were not
filled.  Since they were not filled, the 2x2 quartets had a hole in the
middle (i.e. not flat).  And additionally, because the pads to accept the
chip's bumps were 72um in diameter ... Sierra had to use a "thin copper
foil process" to make such a fine-feature etch.  And that process requires
starting with a starting copper foil process of 12um ... and a part of that
process is to use blind vias for L1-L2 that get copper filled during the
step that plating is done.

The end consequence is that for Sierra's fine-feature process, you "get"
stacked vias along with it.

gwright83, 2020-06-05:
I am closing this (since Kosta and Anita have answered the immediate question); I will open a new issue for Carl's question about whether the Altium database reflects the board as manufactured by Sierra, or a version previous to manufactured one.