FaderOrg/Fader25_board #21: Does the Altium database correspond to the board Sierra built?

gwright83, 2020-06-05:
@kostatro 
@aflynn 
Does the Altium database from the BWRC reflect the final state of the board as manufacturer by Sierra, or did Sierra make changes that were not backported to the BWRC?

The question came up because it appears as if there are thru vias under the IC even in the 
 file fader2_blind_vias_pcb.PcbDoc.  My understanding is that these were plugged so that the under-chip bump attach points would be flat.

aflynn, 2020-06-05:
I'm not sure about the file version.

What actually got fabricated was what I was trying to explain in my
previous email ... nothing was "plugged with epoxy".  That is, the Thru
(unplugged) vias in the first rev were replaced, I believe, by stacked
vias.  When you have stacked vias in the "thin copper foil process", the
L1-L2 via automatically gets *plated* ... i.e. filled with copper (not
plugged with epoxy and plated over).  In the "thin copper foil process"
required for the fine-feature etching, the dielectric between L1 and L2 is
required to be the thinness required for blind vias so that the plating
step by itself will fill the L1-L2 via with copper.

aflynn, 2020-06-05:
You could ask Sierra if you like.  The contact is justinb@protoexpress.com.

gwright83, 2020-06-05:
@aflynn 
Got it.  I will ask them just to be sure.

kostatro, 2020-06-06:
The updated Altium file should reflect the changes required by Sierra. I don't have access to Altium to check it now... It's a good idea to ask Justin.

aflynn, 2020-06-06:
Yes, I have no idea about Kosta's board, but on my last EagleX board,
Sierra did make a change that I approved during the time that their CAM
Shop asks questions, so my file in the repo is not exactly what they
manufactured ... so it happens...

(Their change did not change the functionality of my board, but it
streamlined their process.  The change was to change my Thru vias to a set
of stacked vias, as I mentioned in my previous email.)

gwright83, 2020-06-09:
Picture of the identifying portion of the previous FADER board:
[image]

gwright83, 2020-06-11:
Justin Blakely at Sierra Proto Express sent me a zip of the files used to make the existing board (Sierra job number 151023).  Here it is:
[job_151023.zip](https://github.com/FaderOrg/Fader25_board/files/4766174/job_151023.zip)

This should tell us the solder mask geometry.

gwright83, 2020-06-11:
@caolen 
Hi Carl, you might want to take a look at the files from Sierra, including in the previous comment on this issue.

caolen, 2020-06-11:
Greg, thanks.

I have also forwarded to Pekka and Joe.
Carl

closed 2020-06-18 by gwright83