FaderOrg/Fader25_board #22: 0201 components on Fader board

gwright83, 2020-06-05:
There are some 0201 components on the Fader board.  Is there any reason that they could not be replaced with 0402?  0201 capacitors are especially troublesome, as they have to be operated at a specified DC bias voltage to achieve their rated capacitance.  (This is a consequence of the nonlinearity of the high-epsilon material used in these very small parts.)

If there is no particular reason to use these very small components, we'll substitute 0402s.

aflynn, 2020-06-05:
I never use 0201s, but perhaps Kosta had a reason, I'm not sure.

gwright83, 2020-06-11:
@caolen 
I looked at the datasheet for the TI TPS74201 LDO regulators.  We should replace the 0201 components with 0402s.  If more room is needed for the larger components, the LDO parts can be spaced further apart.

caolen, 2020-06-11:
Greg and all, the 0201 components have been replaced.  So have the through-hole parts.  We’ll review the results at tomorrow’s project call.

Next proposal: move all 34 bottom-side parts to the top side.  We’ll discuss this, too.

Best regards,
Carl
carl.olen@nokia-bell-labs.com<mailto:carl.olen@nokia-bell-labs.com>
+1 908 516-0582
+1 614 406-5443 (mobile)