FaderOrg/Fader25_board #23: 0.1 % Resistors

gwright83, 2020-06-05:
There are some 0.1 % resistors on the Fader board.  Are these necessary, or could they be replaced by more standard 1 % parts.

@caolen 
Carl, if you could send me a pdf of the schematic and a BOM, I could look at the datasheets for the power supply parts to see if there tight tolerance parts are specified by the manufacturers.   I don't have Altium so I can't generate these myself. - Greg

gwright83, 2020-06-17:
@ caolen 
Resistors R1, R3, R5, R6, R8, R40, R72 and R73 are used to set the scale factors for AD5254 digital potentiometers.  The absolute accuracy of these digi-pots is only about 1 percent, so there is no reason to use 0.1 percent resistors here.

R2 and R36 generate a common mode voltage for the clock signal input balun.  This does not require 0.1 percent resistors either.

R11, R12, R13, R14, R15, R16, R17, R18, R19 and R20 are simply pull-up resistors for the NXP PCA9517 I2C bus repeaters.  They require no special accuracy.

Certainly R2 and R36 can be replaced by 1 percent tolerance parts.  We may as well use 1 percent parts for R1, R3, R5, R6, R8, R40, R72 and R73 since the digi-pot absolute accuracy is also about 1 percent.  The pull-ups R11 -> R20 don't even need to be 1 percent tolerance, though there is no harm in using 1 percent parts.

caolen, 2020-06-17:
Greg, thanks.  This gives us more flexibility on these parts.

Best regards,
Carl
carl.olen@nokia-bell-labs.com<mailto:carl.olen@nokia-bell-labs.com>
+1 908 516-0582
+1 614 406-5443 (mobile)

gwright83, 2020-06-18:
Our discussion on today's phone call confirmed that the 0.1 percent resistors have been replaced with 1 percent parts, and this should have no effect on circuit operation.  Closing.

closed 2020-06-18 by gwright83