FaderOrg/Fader25_board #24: Clarification of signal purposes on PCB

mkosunen, 2020-06-26:
We have separated the power supply nets and exposed test points for them.  What would you propose for the following diff pairs?  

We can separate and bring them out as test points, too.  Or, individual SMA ports.  Please let us know.  Our thought is to give more lasting value to this test board as the chip evolves into the future.

- [ ]  LINK_BM_RX_0_P, LINK_BM_RX_0_N
- [ ]   LINK_BM_TX_0_P, LINK_BM_TX_0_N
- [ ]   LINK_BM_RX_3_P, LINK_BM_RX_3_N
- [ ]   LINK_BM_TX_3_P, LINK_BM_TX_3_N
- [ ]   REF_LO_P, REF_LO_N
- [ ]   LINK_BM_RX_2_P, LINK_BM_RX_2_N
- [ ]   LINK_BM_TX_2_P, LINK_BM_TX_2_N
- [ ]   REF_DIG_P, REF_DIG_N
- [ ]   LINK_BM_RX_1_P, LINK_BM_RX_1_N
- [ ]   LINK_BM_TX_1_P, LINK_BM_TX_1_N

 

Also, the diff pair CLK_DIG_OVERRIDE_P, CLK_DIG_OVERRIDE_N is connected through the TC1.5-52TG2+ BALUN to SMA_DIG_CLK.  Are separate test points or SMA connectors desirable for these differential signals?

mkosunen, 2020-06-26:
Also, the diff pair CLK_DIG_OVERRIDE_P, CLK_DIG_OVERRIDE_N is connected through the TC1.5-52TG2+ BALUN to SMA_DIG_CLK.  Are separate test points or SMA connectors desirable for these differential signals?

Consideration: As Far as I recall
LO_override is for down conversion and CLK_DIG_OVERRIDE is a reference for Base band clock divider. Currently they are assumed to be the same signal, but having possibility to use separate sources (additional SMA's)  would not hurt. As alwaysm test poinsts are also nice in case something does not work as intended. However, these are signals at gigaherz frerquencies, and we should sacrifice the signal integrity/matching with these options, So options should be only considered if addiing them is 'safe', as they are not currently mandatory.

@kostatro , any additions or comments to above are highly apreciated.

kostatro, 2020-06-29:
@mkosunen Sorry, something went wrong - I answered a few days ago but now I see that my answer is not here...

I don't have access to Altium, so I need more explanation about the names:
- Are REF_LO_P, REF_LO_N and LO_override the same?
- Are REF_DIG_P, REF_DIG_N and CLK_DIG_OVERRIDE_P, CLK_DIG_OVERRIDE_N the same?
Maybe you can attach a schematic figure of the board to clarify it.

A few more questions:
- What do you mean that "Currently they are assumed to be the same signal"?
- What is the intended digital clock frequency on the SMA connector?

mkosunen, 2020-06-30:
Hi @kostatro, unfortunately I do not have the schematics available.

A) REF_LO_N and REF_LO_P are currently unrouted on chip. The LO_override_N and LO_override_P  are routed to RF_TOP 
B) REF_DIG_P and REF_DIG_N are currently unrouted on chip, CLK_DIG_OVERRIDE_P and CLK_DIG_OVERRIDE_N are routed to RF_TOP

I correct my self above: "As Far as I recall
LO_override is for down conversion and CLK_DIG_OVERRIDE is a reference for Base band clock divider. Currently they are assumed to be the same signal" IS FALSE STATEMENT. 

The remaining questions:
 LO_override, seems to be the  LO input for the the chip (there are no other LO inputs connected) can you confirm that the frequency range is up to 6GHz.
 
 It seems that differential CLK_DIG_OVERRIDE is formed from single ended clock signal with balun and routed to RF_TOP.  D0 you happen remember how it is used inside it? 
All clocks for digital top are driven through RF_TOP, and maximum frequency I used for synthesis is 320 MHz.

kostatro, 2020-06-30:
@mkosunen Regarding the names, I think that the REF signals were intended to be references to the LO and baseband PLLs. We don't have PLLs (right?) so we only use the OVERRIDE inputs.

For the remaining questions:
- LO_override frequency is up to 12GHz. There is an internal division by 2 or by 4, so the RX supports up to 6GHz LO frequency.
- CLK_DIG_OVERRIDE goes to the ADCs and the digital part without dividers in RF_TOP. I think that in fader2 we also used 320MHz in the lab (for the ADC) but I might be wrong.

gwright83, 2020-07-03:
I discussed this with Marko on 3 July 2020 and we agreed that it is unlikely that a future version of the chip will use the same pad configuration (or even the same fab process).  There is no need to break out the currently floating pads, which represent connections to circuit blocks not present in the FADER2.5 chip,  to test points.

Closing this issue.

closed 2020-07-03 by gwright83